EnviroDIY/EnviroDIY_Mayfly_Logger #1: +5V boost efficiency mod

neilh10, 2016-08-18:
From v0.3 of mayfly schematic: If a hardware turn is made, I'm wondering could the +5V generation input L1/C13 be taken before the SPC3819 - ei D3/C7 (J13-2?)
The +5V boost enable of U3 EN could be tied to the 3.3VSW_EN (processor D22 Q2 gate)
This should have no functional impact, but should make the +5V boost circuit more efficient, and guarantee less likelihood of a power glitch on the 3.3V processor power line if a heavy load is present for 3.3V_SW_EN

s-hicks2, 2016-11-29:
Yes, it would be better if the supply for the NCP1402 was on the battery side of the SPX3819 3.3v regulator, however precautions would need to be taken to ensure that the NCP1402 doesn't see more than 6v input, which is why we opted to use the regulated supply instead.  Now that the NCP1402 has been obsoleted, future versions will have a completely different boost circuit.

closed 2016-11-29 by s-hicks2

aufdenkampe, 2018-02-01:
Mayfly v0.5, released in April 2017, has incorporated this feedback.
See https://envirodiy.org/features-of-the-new-mayfly-v0-5/