EnviroDIY/EnviroDIY_Mayfly_Logger #2: +12V generation - NCP1403 opt

neilh10, 2016-08-18:
wrt Mayfly 0.3: If a turn of hardware is done, I wonder would it be possible to conside a layout where by the NCP1403 could be added - possibly as a user option in desoldering the NCP1402 +5V generation.
It seems like the NCP1403 could facilitate generation of  +12V with a reliable current of 20mA, even when the input voltage has dropped to  3.0V, 
The cost would be laying out the board with three optional components 2 resistors, 1cap (RFB1 RFB2 Cc in cct below). Possibly boosting the outcap rated voltage to +15V.
 This components would require one bypass strap to facilitate the current NCP1402-50T1 +5V 

[image]

This could support the large number of +12V sensors, both analog 4-20mA and also with the addition of a RS485 interface the modbus sensors.

s-hicks2, 2016-11-29:
We thought about incorporating a 12v boost option, but since very few end users would actually need it, we decided to leave it off.  We figured it would be easier to generate other voltage on a separate shield or breakout board since it would also require other circuitry like level shifting for the logic and whatever other requirements their signals would need.  But perhaps there's a way to make the layout more modifiable on the next version for people who want to swap out the boost regulator or add an additional one.

aufdenkampe, 2019-03-06:
For reference, neilh10 is referring to the [Modbus-Mafly-WingShield](https://github.com/EnviroDIY/SensorModbusMaster/tree/master/hardware/Modbus-Mayfly_WingShield), which can also easily boost to 9V or 12V.