EnviroDIY/EnviroDIY_Mayfly_Logger #5: Modular Coding for Mayflies

aufdenkampe, 2017-01-23:
This is pasted from an email send by @fryarludwig to @SRGDamia1 about the [SimpleLoggingExample](https://github.com/EnviroDIY/EnviroDIY_Mayfly_Logger/tree/SimpleLoggingExample) and [Modular-Code-Example](https://github.com/EnviroDIY/EnviroDIY_Mayfly_Logger/tree/Modular-Code-Example) feature branches in this repo:

> I've revisited the modular code example, and my original plan for adding multiple values to single sensors would be the wrong approach. A better approach would be more time consuming and would change the current relationship between "sensors" and "dataValues" - I can see this being useful for long-term goals, but is not necessary for the workshop.
> 
> I imagine the SimpleLoggingExample would give you the best results for time committed, given your sensor requirements. This is a single-file example we put together to allow users more control while limiting inherent complexity at the same time.
> 
> To help get you started in the EnviroDIY_Mayfly_Logger.ino file:
> line 37: The config.h file was moved to here - change as needed
> line 74: We have a global sensor variable. Add the variables for the values you want to collect here, and feel free to get rid of the onboard_temperature value.
> line 238: This is the function you'll use to update the variables you created above. Feel free to add a sensor's wake/sleep functions inside this, or you can add those on line 280.
> line 252: Generate your JSON string here for the POST request.
> line 267: This is where we set the time. You mentioned you sync this with the computer, so remove or replace as needed.

aufdenkampe, 2017-01-23:
Once the dust has settled on this sprint, if we start thinking about modularizing the code, we might wan to take a look at the Arduino code here [https://github.com/NorthernWidget/Logger​](https://github.com/NorthernWidget/Logger​). Shannon and I have know the lead developer of that repo, Andy Wickert, for years, and he reached out to us at AGU and in early January about collaborating on a single, modular codebase for Arduino-compatible environmental data loggers.  Andy is that the University of Minnesota, so I'll probably meet with Andy sometime in the next few weeks.

aufdenkampe, 2017-02-20:
@SRGDamia1 just started this effort: https://github.com/EnviroDIY/ModularSensors

SRGDamia1, 2018-09-19:
I think this can safely be closed.

closed 2018-09-19 by SRGDamia1