EnviroDIY/EnviroDIY_Mayfly_Logger #8: GPRSBee class error on "logging_to_EnviroDIY.ino"

aufdenkampe, 2017-02-16:
@SRGDamia1, yesterday 4 of us at LimnoTech tried out https://github.com/EnviroDIY/EnviroDIY_Mayfly_Logger/blob/master/examples/logging_to_EnviroDIY/logging_to_EnviroDIY.ino by making modifications in separate personal branches.

Unfortunately, I couldn't get the sketch to Verify, even though I have my libraries updated to the most recent. On my two different laptops (both macOS 10.12.3), I got a `'class GPRSbeeClass' has no member named 'addHTTPHeaders'` error on line 477. However my 3 colleagues, all running windows, had no problem verifying their sketches and now one is logging to Data.EnviroDIY.org.

Here is the full error:

```
Arduino: 1.8.0 (Mac OS X), Board: "EnviroDIY Mayfly 1284p"

/Users/aaufdenkampe/Documents/Arduino/EnviroDIY_Mayfly_Logger/examples/logging_to_EnviroDIY/logging_to_EnviroDIY.ino: In function 'int postDataGPRS(bool)':
logging_to_EnviroDIY:477: error: 'class GPRSbeeClass' has no member named 'addHTTPHeaders'
     gprsbee.addHTTPHeaders(header);
             ^
logging_to_EnviroDIY:478: error: 'class GPRSbeeClass' has no member named 'addContentType'
     gprsbee.addContentType(F("application/json"));
             ^
Multiple libraries were found for "SD.h"
 Used: /Users/aaufdenkampe/Documents/Arduino/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
exit status 1
'class GPRSbeeClass' has no member named 'addHTTPHeaders'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Any ideas?

SRGDamia1, 2017-02-16:
It really looks like your library is not up to date.  Given you're also getting an error with finding double libraries for SD.h (which is a built-in that I didn't touch), I think there might be something goofy going on with how your software is finding your libraries.  Are you sure you restarted the IDE after updating?  You definitely did not update the library with the IDE by recloned from git, right?  Can you post the verbose output?

SRGDamia1, 2018-09-19:
This example should just be removed; it's pretty "ancient."

brianjastram, 2019-02-22:
I'm getting a similar GPRSBee error at the end of the below copied verbose report.
I just tried the BPRSbee-master library I extracted from https://github.com/SodaqMoja/GPRSbee.
Can you tell what I'm missing?

Arduino: 1.8.8 (Windows 7), Board: "EnviroDIY Mayfly 1284p"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\brian.jastram\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\brian.jastram\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries U:\My Documents\Arduino\libraries -fqbn=EnviroDIY:avr:envirodiy_mayfly -ide-version=10808 -build-path C:\Users\BRIAN~1.JAS\AppData\Local\Temp\arduino_build_637140 -warnings=none -build-cache C:\Users\BRIAN~1.JAS\AppData\Local\Temp\arduino_cache_574906 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=C:\Users\brian.jastram\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.0.1-arduino5.path=C:\Users\brian.jastram\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino5 -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose U:\My Documents\Arduino\Sketches\EnviroDIY_Mayfly_Logger_Feb22\EnviroDIY_Mayfly_Logger_Feb22.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\brian.jastram\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\brian.jastram\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries U:\My Documents\Arduino\libraries -fqbn=EnviroDIY:avr:envirodiy_mayfly -ide-version=10808 -build-path C:\Users\BRIAN~1.JAS\AppData\Local\Temp\arduino_build_637140 -warnings=none -build-cache C:\Users\BRIAN~1.JAS\AppData\Local\Temp\arduino_cache_574906 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=C:\Users\brian.jastram\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.0.1-arduino5.path=C:\Users\brian.jastram\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.0.1-arduino5 -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose U:\My Documents\Arduino\Sketches\EnviroDIY_Mayfly_Logger_Feb22\EnviroDIY_Mayfly_Logger_Feb22.ino
Using board 'envirodiy_mayfly' from platform in folder: C:\Users\brian.jastram\AppData\Local\Arduino15\packages\EnviroDIY\hardware\avr\1.0.0
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=8000000L -DARDUINO=10808 -DARDUINO_AVR_ENVIRODIY_MAYFLY -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Users\\brian.jastram\\AppData\\Local\\Arduino15\\packages\\EnviroDIY\\hardware\\avr\\1.0.0\\variants\\mayfly" "C:\\Users\\BRIAN~1.JAS\\AppData\\Local\\Temp\\arduino_build_637140\\sketch\\EnviroDIY_Mayfly_Logger_Feb22.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega1284p -DF_CPU=8000000L -DARDUINO=10808 -DARDUINO_AVR_ENVIRODIY_MAYFLY -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Users\\brian.jastram\\AppData\\Local\\Arduino15\\packages\\EnviroDIY\\hardware\\avr\\1.0.0\\variants\\mayfly" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "C:\\Users\\BRIAN~1.JAS\\AppData\\Local\\Temp\\arduino_build_637140\\sketch\\EnviroDIY_Mayfly_Logger_Feb22.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega1284p  [... cut ...]
Using library Sodaq_RTCTimer at version 1.0.0 in folder: U:\My Documents\Arduino\libraries\Sodaq_RTCTimer 
Using library EnviroDIY_DS3231 at version 1.3.2 in folder: U:\My Documents\Arduino\libraries\EnviroDIY_DS3231 
Using library EnviroDIY_PCInt_PCINT0 at version 1.2.1 in folder: U:\My Documents\Arduino\libraries\EnviroDIY_PCInt_PCINT0 
Using library GPRSbee-master at version 1.9.2 in folder: U:\My Documents\Arduino\libraries\GPRSbee-master 
exit status 1
'class GPRSbeeClass' has no member named 'addHTTPHeaders'

SRGDamia1, 2019-02-22:
Just to warn you, this example is NOT what I'd currently recommend using.  I would (strongly) suggest switching to the [ModularSensors ](https://github.com/EnviroDIY/ModularSensors) library to send data to enviroDIY.

To use this sketch you need to use branch v1.2_hacked of the EnviroDIY fork of the GPRSBee library:  https://github.com/EnviroDIY/GPRSbee/tree/v1.2_hacked.  That version was the last one to work properly on the Mayfly, and then was modified to allow the authentication headers needed to communicate with the EnviroDIY Data Portal.

SRGDamia1, 2019-02-22:
@brianjastram  If you're using the Arduino IDE, I would also recommend moving away from that in favor of PlatformIO