EnviroDIY/EnviroDIY_Mayfly_Logger #18: PWM ports

WeatherNut, 2018-08-29:
I have recently been trying to connect multiple analog sensors to the mayfly board using a 16-channel analog multiplexer. I have reached a little hurdle however. The PWM ports are clearly labeled on the arduino uno boards, while the mayfly they are not. Does anyone now which digital ports are PWM ports?

SRGDamia1, 2018-08-29:
Should be pins D4, D5, (Timer 1 - 16 bit) D6, D7, (Timer 2 - 8 bit) D11, D12, (Timer 0 - 8 bit) D14, D15 (Timer 3, 16 bit).

closed 2018-08-29 by SRGDamia1

aufdenkampe, 2018-08-30:
@WeatherNut, the Mayfly is built on the [ATmega1284p](http://ww1.microchip.com/downloads/en/DeviceDoc/doc8059.pdf) chip, so digging into those specs is an important place to start.

Then, to translate to the numbering scheme printed on the Mayfly, you'll need to look at the Mayfly schematics here: https://github.com/EnviroDIY/EnviroDIY_Mayfly_Logger/tree/master/hardware

@SRGDamia1, was there an easier source of info for this than the 380 page [ATmega1284p](http://ww1.microchip.com/downloads/en/DeviceDoc/doc8059.pdf) spec sheet?

SRGDamia1, 2018-09-03:
Actually, the spec sheet wouldn't have told you which "Arduino" digital pin numbers could do PWM, only that all the OC pins from all timers were capable and which physical pins those were.  You need to look at the [pins_arduino.h](https://github.com/EnviroDIY/Arduino_boards/blob/master/envirodiyboards/variants/mayfly/pins_arduino.h) for the Mayfly to verify the assignment of the "Arduino" digital pin numbers to physical pins.  (They are not the same, and in fact the assignments on the Mayfly match the Mbili but are different from most other 1284p boards.)  The block with "digital_pin_to_timer_PGM" (line 208) its the actual assignment of physical timer pins to digital "Arduino" pins.

SRGDamia1, 2018-09-03:
The timer OC pins are also shown on the Mayfly schematic, along with their digital pin numbers:  https://raw.githubusercontent.com/EnviroDIY/EnviroDIY_Mayfly_Logger/master/hardware/mayfly_v0p5b_schematic.png

I don't know if there was ever another place where the PWM capable pins are specifically flagged.  I've never tried any pulse width modulation applications so I had to dig around a bit to figure out which pins could do it.