EnviroDIY/EnviroDIY_Mayfly_Logger #21: sleep mode current

neilh10, 2018-11-02:
Just checking on mA consumed during sleep - 
for the current 'develop' branch using logging_to_EnviroDIY.ino, on sleep the current reduces to about 3.5mA after EnviroDIYLogger.systemSleep() is executed. 
This is with a barebones Mayfly0.5b  and only one sensor ProcessorStats_SampleNumber()  in variableList[], no SD card, no Xbee or anything. Just the rtc coin battery added to the board.
I'm expecting to get to about 0.5mA as per  @s-hicks2  https://www.envirodiy.org/topic/low-power-project/
With Sensors sampling and Green and Red led its about 10.5mA

If I comment out the real work of sensor update - and just leave 
EnviroDIYLogger.checkInterval() - 
EnviroDIYLogger.systemSleep() 
then in one case it went to sleep and consumed 0.5mA - hurrah.

However on repeating the same exercise above, it didn't  reduce the current - but stayed at 3.5mA
So it could be that I just got lucky once - and it doesn't matter whether its actually running the sensors update or not.

I'm measuring current as taken for the LiIon battery with a good digital multimeter with a mA range (and also uA range). 
I've modified LiIon battery extension cable so that it works for a running mayfly.
I power up from cold by plugging in the LiIon cable
  
I can see EnviroDIYLogger.systemSleep() has a lot of good code to do a powerdown, but I'm not getting there. 

Just wondering if I'm missing a switch, or any suggestions on how to measure sleep current.  Thanks

SRGDamia1, 2018-11-02:
I've honestly never tested the sleep (or wake) current draw.

neilh10, 2018-11-02:
Thanks - just checking and putting it out there to see if anybody has any insights.
Its a difficult challenge as its like a leaky water line, so many place for current/uA to leak out.  
The Mayfly0.5 is a bit touchy on a longer battery cable and having a Digital Meter in the path - it was resetting - so needed to make the cable with a low impedance capacitor at the JP1/JP2 end.
There is a sketch that I can use to verify the base hardware and make sure that it is capable of getting that low - https://www.envirodiy.org/mayfly/software/sleeping-mayfly-logger-example/   
I seemed to get to 0.5mA once, so might be something like a floating input.  
There is a good structure of code to work with :).

SRGDamia1, 2018-11-02:
@aufdenkampe - want to send me one?

SRGDamia1, 2018-11-02:
Almost all of the work I've done to minimize power draw has been in working to minimize the sensor-on time.  The only "special" thing I have coded in to reduce the draw while actually sleeping is to force the I2C pins to go low in sleep to reduce parasitic draw.  On the processor, I'm disabling the on-board ADC, the brown-out detector, the various internal clocks, and going into "SLEEP_MODE_PWR_DOWN", which is the lowest power use setting.

neilh10, 2018-11-02:
I could write up a method for doing sleep power measurements.  
I think the goal needs to be clear and its a technically difficult challenging area . 
I would think the goal is that there is an easy  method to get good enough power for any release of software with some low cost (time and materials) methods.
Its technically challenging as there is a wide dynamic range of power demand and distributed  - the dynamic range is a technical problem.  That is when some Cellular modems are transmitting they can take 1000mA for short bursts. 
However for sleep measurements, need to be able to determine that the Mayfly power is about 0.5mA - or the measurement accuracy needs to be somewhere round +-/0.25mA (eg 0.1mA would be fine)
The problem is that this can only be quickly and accurately determined from the battery line.
If it is measured from the external line, then the LiPo needs to be disconnected so that it isn't taking current, and then I see the LiPo charger flashing LED4. I can experiment a bit more so if there is a stable condition with low quiescent for the LiPo charger +LED.
The MakerHawk UM25C USB Tester, looks to be excellent with at least a resolution of 0.1mA, but states its accuracy is 1%, which at 5.0000A is +-0.0500A or 50mA - so need to check if can actually measure accurately to at least 0.5mA.   It also looks to have a power accumulation register which would be good.
I have got a couple of lower cost USB testers and they only have a +/-10mA resolution.
I'll get a UM25C to try it.

For a LiPo cable current measurement if its of interest - it would be about having a Digital Voltmeter with 0.1mV resolution - and then a modified battery extension cable with a 0.1ohm in series most of the time. For a current pulse of 1A this would drop 100mV - which I think would be OK, but needs to be verified.
For sleep power measurements, a push button momentary switch would put in a 10 ohms resistor, and then a sleep current of 0.5mA would be 5mV.  Just a thought. I can could make up a few cables and distribute them if anybody is interested and has the Digital Voltmeter.

neilh10, 2018-11-03:
If I use the 
https://www.envirodiy.org/mayfly/software/sleeping-mayfly-logger-example/
and for platformIO I build against
https://github.com/EnviroDIY/Libraries.git#platformio
then when it sleeps I get 0.40mA. This is with a ScanDisk Ultra 16G inserted.
When its running, briefly writing to the SD card it peaks 7.9mA
I visually can't see the differences between sleeping-mayfly and logging_to_EnviroDIY.ino at this point, but I'm looking at it section by section.

neilh10, 2018-11-05:
I'm trying another approach that seems to be working , using a current monitor [Adafruit/904](https://www.adafruit.com/product/904) which has INA219B, with the I2C on a separate Mayfly board.
The results are verified with my Multimeter mA measurements.
The INA219B has 0.1mA resolution for series impedance of 0.01ohms, and a range +-400mA
In the picture the LiIon is +ve is connected to the INA219B, the Mayfly0.4#1 with the USB cable polls the I2C and reports current
The Mayfly0.5b#2 (with screwdriver & Xbee) has its power solely provided by the Rbat. The J2/FTDI monitoring cable has its power and DTR pins clipped so its only monitoring, and not providing power or causing a tty reset.
[image]

The Mayfly#1 running a modified
//https://github.com/adafruit/Adafruit_INA219/blob/master/examples/getcurrent/getcurrent.ino

With no Xbee, and Mayfly#2 running a simple program writing to the SD card every minute
https://www.envirodiy.org/mayfly/software/sleeping-mayfly-logger-example/

the results from Mayfly#1 are with polling about every 2seconds what you would expect - the multimeter reported 0.38mA - which the INA219B is limited to 0.1mA resolution with no rounding of measurements.

---Ina219(BUILD_TIMESTAMP 1541382006)
Reporting Power and current with INA219 0-400mA ...
mA:   0.30, V: 4.13, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   7.20, V: 4.11, mW:   29
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   7.10, V: 4.11, mW:   29
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.12, mW:    2
mA:   0.30, V: 4.11, mW:    2
mA:   0.30, V: 4.12, mW:    2
-----
I have a spare adafruit.com/product/904  and cables so I could make up a similar setup tomorrow and share it...
I speculatively ordered/Digikey on Fri the following (hopefully arriving tomorrow) 
two 3650	INA219 FEATHERWING
one 2796	EVAL BRD FEATHER M0 ADALOGGER

aufdenkampe, 2018-11-05:
@neilh10, thanks for all these tests and ideas. I'll see what I can do to replicate your setup, and then we'll see if we can get @SRGDamia1 setup too.

SRGDamia1, 2018-11-06:
@aufdenkampe - I already have the feather M0, I'd just need the the featherwing and cables.

aufdenkampe, 2018-11-06:
@gcutrell, I think it would be really helpful to us to set up a high-frequency power recording system similar to the one that @neilh10 set up in his https://github.com/EnviroDIY/EnviroDIY_Mayfly_Logger/issues/21#issuecomment-435734443 above, which uses an [Adafruit INA219 High Side DC Current Sensor Breakout](https://www.adafruit.com/product/904) and an extra Mayfly. What do you think?

I know you've had success with the [MakerHawk UM25C USB Tester](https://www.amazon.com/MakerHawk-Bluetooth-Current-24-000V-Multimeter/dp/B07C9DQ4F9), but I think this might be able to provide a more accurate power readout for our purposes of figuring out what portions of the code are using the most power, because this enables us to configure the Mayfly exactly as we would for a field deployment.

neilh10, 2018-11-06:
Interesting find. It has three sensors - but turns out that the channels aren't as sensitive. 
The INA3221 1-LSB size for shunt voltage is +-40uV.  Actually mA measurement will depend on series shunt resistance and couldn't find that on their description, but nice to have resolution at about 0.1mA
For the sleep measurements, the 1LSB step size and Offset voltage are critical. The offset voltage determines how accurate it can measure 0.000V
The INA219B has 1-LSB +-10uV resolution, and also for low offset+/- 10uV offset - meaning that it would only error about 0.00000 by +-10uV.  
http://www.ti.com/lit/an/sboa194/sboa194.pdf compares the different TI devices. 
A board using  INA233 would be interesting as its sensitivity is 2.5uV and offset 2.5uV and contains a power Accumulator - ie can calc mAHrs - that would be useful to measure used mAHrs instead of just a digital sample at one point in time. 
The one other chip I've been keeping an eye on is Microchip PAC1720 dual (PAC1710 single) monitor,it is  low cost with voltage sense at 1LSB +-5uV. I have it designed in on a MKL27Z (cortex-M0) board - but not written a driver for it.

neilh10, 2018-11-28:
Great - would it work to make up battery cable so you can insert it inline with the battery?. I believe that is the easiest way of measuring sleep current.
Alternatively you can clip a battery cable's +V and insert it inline.
It also needs a groove 4wire cable.
https://github.com/neilh10/ModularSensors/wiki/Test-equipment-cable-monitor-LiIon
The code I've been using is on my branch "test_nh1"
https://github.com/neilh10/ModularSnsrsNh/tree/test_nh1/sensor_tests/Adafruit_INA219
I've also included it as a sensor (I keep meaning to clean it up and submit it) 

https://github.com/neilh10/ModularSensors/tree/work_nanolevel/examples/logging_to_EnviroDIY/lib/ModularSensors/src/sensors/TiIna291.h/cpp