EnviroDIY/EnviroDIY_Mayfly_Logger #23: Upgrade solar charge controller.

aufdenkampe, 2019-03-06:
The Mayfly's Lipo battery solar charge controller chip ([MCP73831](https://www.microchip.com/wwwproducts/en/en024903#additional-features)) has a max output of 500 mA, which is about the 550 mA peak output of the [Voltaic's 3.5W solar panel](https://www.voltaicsystems.com/3-5-watt-panel) that we often use for our power-hungry sensors (i.e. YosemiTech turbidity with a brush) along with Adafruit's [2-cell 4400mAh](https://www.adafruit.com/product/354) or [3-cell 6600mAh](https://www.adafruit.com/product/353) Lithium Ion Battery Packs. This means that we can't increase the solar panel size beyond this for more power-hungry applications (such as several sensors with brushes).

It would be great to have a slightly higher rated and potentially more efficient charge controller. 

One option might be the [MCP73871](https://www.microchip.com/wwwproducts/en/en536670) controller that is used in the [Adafruit Solar Lithium Ion/Polymer charger - v2](https://www.adafruit.com/product/390) that can be modified to 1A charging and gets high marks in this review: https://www.voltaicsystems.com/blog/lithium-ion-charge-controller/.

Not sure if there are other pros/cons. Is there value at going with a MPPT (maximum power point tracker) charge controller?

everhamme, 2019-03-07:
Per most recommendations I've seen true MPTT is overkill and expensive.  Since our incoming voltage from solar is near 6V it's not too far off from the required charge voltage.  If we switch to 12v panels than Mptt might be necessary.

Also note that for max efficiency we should limit the charge current of the charger (via custom resistor) to match that as close to possible of the expected conditions in ther field or based on expected max output of panel..... so size charging rate to panel size.   It's documented on the adafruit page in great detail.    https://learn.adafruit.com/usb-dc-and-solar-lipoly-charger/using-the-charger

_____________________________________
Ed Verhamme
Project Engineer
501 Avis Drive, Ann Arbor, MI 48108
Office: 734-332-1200<tel:734-332-1200> Direct/Cell: 734-681-0577<tel:734-681-0577>
Email: everhamme@limno.com<mailto:everhamme@limno.com>
http://twitter.com/eddiegreatlakes
LimnoTech           www.limno.com<http://www.limno.com/>
Water Environment | Scientists Engineers

neilh10, 2019-03-07:
Great idea to move to 1A - that could mean a 8Ahr battery could be recharged in a solar day.

For an easy option to move to a higher voltage a 3pin plug-in switcher could be used. I have been thinking about could it built into the cable.  Its 3-pins are centered on 0.1" - so a fairly small footprint. It would require a different polarized 2pin connector for the higher solar voltage.   
(For the higher W solar panels an 18V typical can get to 30V on a cold day and no charge being taken) 
OKI-78SR-5/1.5-W36-C Vin=36V, Vout 5V@1.5A
https://www.digikey.com/products/en?keywords=811-2196-5-ND  $4.50

The MCP73871 is nice for its power-path switcher. That means that for a fully discharged LiIon (dead low voltage disconnect to processor) with weak sunshine (high impedance solar) - it will provide power to the uP before it starts charging the LiIon  - the uP needs to know that it can't place a heavy demand on the battery. There would be a benefit for a direct reliable LiIon battery V instead of the current USB V. ( I'm already testing this code for myself).

The current Mayfly MCP charger trickle charge a fully discharged battery till the LiIon voltage rises enough before allowing the full charge current. That of course can be annoying to a person sitting in front of the system trying to determine if its healthy as it won't come up for some time. 

There are some LiIon chargers that enable the charge current to be programmed by toggling the enable line - but I forget which ones now.

aufdenkampe, 2019-03-07:
@everhamme and @neilh10, thanks for your encouragement and input! (also, @everhamme gets credit for pointing me to the blog post from Voltaic!).

I really like the idea of keeping future versions of the Mayfly to be flexible, for a broad range of power needs. I also like the idea of sticking within the range of [Voltaic's small 6V solar panels](https://www.voltaicsystems.com/shop-gear/categories/solar-panels), which include 1, 2, 3.5, 6, and 9 W sizes. And I like the idea of sticking with the same JST connector that is broadly used in the DIY electronics world.

So maybe the upgrade is to the MCP73871 controller, but with the ability to switch between 0.5A or 1.0A charging, using something like the 3-pin switching jumpers near the Mayfly's grove ports?