EnviroDIY/EnviroDIY_Mayfly_Logger #31: Xbee Decoupling capacitor

neilh10, 2021-10-20:
Cellular modules recommend a low ESR ceramic decoupling capacitor close to the power pins, with the lowest impedance current loop path. 
For Maylfly 1.0A3 there doesn't appear to be a large decoupling capcitor between BEE_VCC and GND, though perhaps the path through the regulator to the regulator's 10uF and low impedance LiIon battery wroks
Can a largish ceramic capacitor (100uf) be applied to Xbee pins. 
I'll post references to data sheets, but I have limited access this afternoon.

s-hicks2, 2021-11-03:
Sufficient power filtering capacitors are included on the EnviroDIY LTE sim7080 cell module, so no additional caps were added to the Mayfly 1.0.  But since some users might want to use the Digi brand LTE boards directly on the Mayfly, filtering cap(s) will be added to future versions.  Short term tests of the Digi LTE bee directly plugged into the Mayfly with no LTEbee adapter (making appropriate changes to the solder jumpers on the back of the Mayfly 1.0) showed no problems with battery power input.

aufdenkampe, 2021-11-03:
@s-hicks2, thanks for creating a new cellular modem for the EnviroDIY community. 

I agree with @neilh10 that the Mayfly v1 needs to support the Digi Xbee3 LTE-M modems -- both the older ones that we are already using and also the major revision that they are currently developing -- in addition to other future modems. Therefore, I think it is important to add that capacitor chain that @neilh10 displayed above.

neilh10, 2022-03-30:
The Mayfly 1.1 Xbee socket has a good capacitor decoupling for supporting an Digi Xbee LTE. Many thanks

closed 2022-03-30 by neilh10